zephyrproject-rtos/zephyr #95811: Conflict Between POSIX Networking and NET_HOSTNAME_UNIQUE_UPDATE in Zephyr v3.7.1

PragatiGarg-eaton, 2025-09-10:
### Describe the bug

@cfriedt, @dleach02, I am trying to use the NET_HOSTNAME_UNIQUE_UPDATE feature in Zephyr 3.7.1 to support runtime hostname updates. However, this configuration is no longer viable due to a conflict introduced by automatic dependency selection in Zephyr 3.7.1

### **Issue Summary:**
In Zephyr 3.6, the following configuration worked as expected:
```
CONFIG_POSIX_API=y
CONFIG_NET_HOSTNAME_UNIQUE_UPDATE=y
```

- Hostname updates were successfully triggered, and there were no dependency conflicts.

- In Zephyr 3.7.1, enabling` CONFIG_POSIX_API=y`, auto-selects `CONFIG_POSIX_NETWORKING=y`. This in turn force-selects `CONFIG_NET_HOSTNAME_DYNAMIC=y,` which conflicts with `CONFIG_NET_HOSTNAME_UNIQUE_UPDATE=y.`

#### Conflict Chain:
 `CONFIG_POSIX_API=y`
  ↓
`CONFIG_POSIX_NETWORKING=y`
  ↓
`CONFIG_NET_HOSTNAME_DYNAMIC=y`
  ↓
// CONFLICTS with `CONFIG_NET_HOSTNAME_UNIQUE_UPDATE=y`

#### **Queries:**
I would like to continue using NET_HOSTNAME_UNIQUE_UPDATE for hostname customization and runtime updates. Please advise on:

1. Whether there is a supported way to override or decouple NET_HOSTNAME_DYNAMIC from CONFIG_POSIX_NETWORKING.
2. If not, can a patch or configuration workaround be provided to allow coexistence of POSIX_API and NET_HOSTNAME_UNIQUE_UPDATE?
3. If this change is intentional, could documentation be updated to reflect the new limitations and suggest migration paths?

cc: @nashif 

### Regression

- [ ] This is a regression.

### Steps to reproduce

CONFIG_POSIX_API=y
CONFIG_NET_HOSTNAME_UNIQUE_UPDATE=y

### Relevant log output

```shell
warning: NET_HOSTNAME_DYNAMIC (defined at subsys/net/Kconfig.hostname:20) has direct dependencies !NET_HOSTNAME_UNIQUE_UPDATE && NET_HOSTNAME_ENABLE && NETWORKING with value n, but is currently being y-selected by the following symbols:
 - POSIX_NETWORKING (defined at lib/posix/options/Kconfig.net:5), with value y, direct dependencies NETWORKING (value: y), and select condition NETWORKING (value: y)

error: Aborting due to Kconfig warnings
```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

- Zephyr Version: 3.7.1
- Board: nucleo_h743zi
- Toolchain: Zephyr SDK 0.16.9
- Build System: CMake + west
- Host OS: Ubuntu 22.04 LTS / Windows
- Networking Stack: POSIX Networking enabled via CONFIG_POSIX_API=y

cfriedt, 2025-09-13:
@PragatiGarg-eaton - thanks for checking on main too.

I looked into it as well, and it might not be a circular dependency - potentially just conflicting Kconfig options.

It has to do with limits.

POSIX has 2 required limit values that come into play here:

* `_POSIX_HOST_NAME_MAX`
  * a "minimum value" (i.e. constant, unmodifyable)
  * must be equal to 255
* `HOST_NAME_MAX`
  * runtime invariant (i.e. compile-time constant, but modifyable)
  * should be at least `_POSIX_HOST_NAME_MAX`
  * configured in Zephyr via `CONFIG_POSIX_HOST_NAME_MAX`

The latter can be omitted if the implementation requires the value to be queried at runtime via `sysconf(_SC_NAME_MAX)`. Runtime queries are costly though in firmware, so we should probably just use the constant.

In the POSIX library, `CONFIG_POSIX_HOST_NAME_MAX` is defaulted to `CONFIG_NET_HOSTNAME_MAX_LEN`.

However, `CONFIG_NET_HOSTNAME_MAX_LEN` depends on `CONFIG_NET_HOSTNAME_DYNAMIC`.

So the only reason that POSIX selects `CONFIG_NET_HOSTNAME_DYNAMIC` is because that is the only way to coerce Zephyr into defining a limit.

The solution, I think, would need to be in the networking subsystem. cc @jukkar @rlubos. My suggestion is below.

Technically, `CONFIG_NET_HOSTNAME_MAX_LEN` is just a number, and does not strictly depend on `CONFIG_NET_HOSTNAME_DYNAMIC`.

Firmware can be conditionally compiled to either make it possible to update the hostname or make it impossible (and return an error).

Build-time assertions can also be used to ensure that various assumptions are true. For example, if we do not have `CONFIG_NET_HOSTNAME_ENABLE=y`, then the hostname is statically set to "zephyr". In that case, I would expect `CONFIG_NET_HOSTNAME_MAX_LEN=6`. Even though we cannot update it, it's still a valid upper bound.

A good starting would be the change below.
```diff
diff --git a/subsys/net/Kconfig.hostname b/subsys/net/Kconfig.hostname
index eb082d93476..7eb0c3d3eff 100644
--- a/subsys/net/Kconfig.hostname
+++ b/subsys/net/Kconfig.hostname
@@ -9,6 +9,14 @@ config NET_HOSTNAME_ENABLE
          This is used for example in mDNS to respond to <hostname>.local
          mDNS queries.
 
+config NET_HOSTNAME_MAX_LEN
+       int "The maximum allowed hostname length"
+       # This corresponds to "zephyr"
+       default 6 if !NET_HOSTNAME_ENABLE
+       default 63
+       help
+         This will set the number of bytes allocateed for the hostname.
+
 if NET_HOSTNAME_ENABLE
 
 config NET_HOSTNAME
@@ -24,14 +32,6 @@ config NET_HOSTNAME_DYNAMIC
          This will enable the net_hostname_set() function. NET_HOSTNAME
          will be used as default hostname.
 
-config NET_HOSTNAME_MAX_LEN
-       int "The maximum allowed hostname length"
-       depends on NET_HOSTNAME_DYNAMIC
-       range 1 63
-       default 63
-       help
-         This will set the number of bytes allocateed for the hostname.
-
 config NET_HOSTNAME_UNIQUE
        bool "Make hostname unique"
        help
diff --git a/subsys/net/hostname.c b/subsys/net/hostname.c
index 7b85b4782a8..3fe8454d8d9 100644
--- a/subsys/net/hostname.c
+++ b/subsys/net/hostname.c
@@ -18,7 +18,9 @@ LOG_MODULE_REGISTER(net_hostname, CONFIG_NET_HOSTNAME_LOG_LEVEL);
 #include <zephyr/net/net_mgmt.h>
 #include <zephyr/logging/log_backend_net.h>
 
-static char hostname[NET_HOSTNAME_SIZE];
+BUILD_ASSERT(CONFIG_NET_HOSTNAME_MAX_LEN > 0, "NET_HOSTNAME_MAX_LEN must be a positive value");
+
+static char hostname[NET_HOSTNAME_SIZE] = CONFIG_NET_HOSTNAME;
 
 static void trigger_net_event(void)
 {
```

@PragatiGarg-eaton - are you up for making a PR? 

If so, there will also be some code changes needed, and ideally additional test configurations (`tests/posix/net/testcase.yml`) and cases (`tests/posix/net/src/gethostname.c`) to ensure that the necessary coverage is reached.

PragatiGarg-eaton, 2025-09-13:
Thanks for the analysis and suggestions. Yes I’ll be happy to take this on. @cfriedt I’ll create the PR, include the required code changes, and update the test config and cases to ensure coverage. I’ll add you as reviewer once it's up.

cfriedt, 2025-09-14:
I have some housekeeping for `gettimeofday()` in the works as well, which should hopefully simplify your PR @PragatiGarg-eaton 
https://github.com/zephyrproject-rtos/zephyr/pull/95939

PragatiGarg-eaton, 2025-10-15:
> [@PragatiGarg-eaton](https://github.com/PragatiGarg-eaton) - are you able to make the PR against Zephyr's `main` branch?
> 
> I can review the PR to the fork as well, but it should go upstream

Yeah, I'll raise it for upstream.