zephyrproject-rtos/zephyr #95938: TF-M: Nordic: unable to specify secure read ranges in-tree

JordanYates, 2025-09-13:
### Describe the bug

It is no longer possible to specify extra secure memory regions that the NS application has permissions to read through the [`tfm_read_ranges.h`](https://github.com/zephyrproject-rtos/zephyr/blob/main/modules/trusted-firmware-m/nordic/include/tfm_read_ranges.h) header.

Two part cause:
1. `tfm_read_ranges.h` was renamed to `tfm_platform_user_memory_ranges.h` in https://github.com/zephyrproject-rtos/trusted-firmware-m/commit/57b33427d15fbbb966ee3991c1ae4471364259b4, but the in-tree file was not updated (see https://github.com/zephyrproject-rtos/zephyr/blob/ced0cc307dc0482d29002e228a1cf8403005b539/modules/trusted-firmware-m/nordic/include/tfm_read_ranges.h#L7). 
2. New standalone variants of `tfm_platform_user_memory_ranges.h` were introduced in TF-M in https://github.com/zephyrproject-rtos/trusted-firmware-m/commit/9e3750992106e27b9501b9ddb39446effc3881c2, unconditionally on the include path (for the purposes of Zephyr, `NRF_DIR` is the Nordic downstream)

As a result of 2, fixing 1 doesn't change anything, since the TF-M repo files are still prioritized when including files.

### Regression

- [x] This is a regression.

### Steps to reproduce

Go to `tfm_read_ranges.h`, add `#error not compiled`.
Compile a TF-M enabled Nordic application.
Application will compile fine.

### Relevant log output

```shell

```

### Impact

Functional Limitation – Some features not working as expected, but system usable.

### Environment

 - Zephyr v4.2

Vge0rge, 2025-09-19:
Hey @JordanYates, thank you for the bug report! Your are right that this is an issue. It seems that with the current logic it makes sense to remove the tfm_read_ranges.h all together from Zephyr and only keep it in TF-M. Does that make sense to you?

If so I can open a PR to remove it.

JordanYates, 2025-09-19:
> Hey [@JordanYates](https://github.com/JordanYates), thank you for the bug report! Your are right that this is an issue. It seems that with the current logic it makes sense to remove the [...]

No, that is a regression in functionality, as specified in the issue. The point of the header is to enable updating the ranges without needing to go through TF-M.