zephyrproject-rtos/zephyr #95973: Integrating HAL_XHSC Version Information into west.yml

TrackingRobot, 2025-09-15:
### Problem Description

A few months ago, I submitted a request for creating the external module repository [hal_xhsc](https://github.com/zephyrproject-rtos/zephyr/issues/86804). This request has been approved by TSC and the repo has been created. 
Now we are working on [PR#94044](https://github.com/zephyrproject-rtos/zephyr/pull/94044). That PR needs to modify west.yml to add the information of hal_xhsc. However, neither of us has permission to change west.yml. 
Therefore, we have opened this issue for the request to add the information of hal_xhsc.

### Proposed Change (Summary)

We hope to add the information for zephyrproject-rtos/hal_xhsc into west.yml, which is necessary to compile the Zephyr project for XHSC SoCs.

### Proposed Change (Detailed)

Add the following content in the appropriate place (currently line 227) in the file west.yml.
```
      groups:
        - hal
    - name: hal_xhsc
      revision: a84e04900616f68097d80cda2e89eaa8af3afadd
      path: modules/hal/xhsc
```

henrikbrixandersen, 2025-09-15:
You need to open a pull request (PR) for making changes to any file in the repo. Please include the changes to west.yml in the same PR introducing code depending on the XHSC HAL.

- https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request
- https://docs.zephyrproject.org/latest/contribute/index.html

TrackingRobot, 2025-09-16:
> You need to open a pull request (PR) for making changes to any file in the repo. Please include the changes to west.yml in the same PR introducing code depending on the XHSC HAL.
>
>     * [...]

Thanks for your suggestion. The reason I open this issue for changing west.yml is that the CI check indicating that I havn't acess to change that file. Now, I will follow your suggestion go on the PR#94044 that including changes about west.yml.