zephyrproject-rtos/zephyr #95977: RFC: Introduce Zephyr driver for Local Interconnect Network (LIN)

thenguyenyf, 2025-09-15:
### Problem Description

Currently, Zephyr doesn't support any Local Interconnect Network (LIN) software stack. We are trying to add a LIN communication stack following the OSI model, referring to LIN Specification Pack Rev2.1 (published by LIN CiA). The first step is to add the LIN physical layer and part of the data link layer as a Zephyr device driver.

### Proposed Change (Summary)

Introduce the LIN driver subsystem, under `include/zephyr/drivers/lin.h` and `drivers/lin`

### Proposed Change (Detailed)

See https://github.com/zephyrproject-rtos/zephyr/pull/95975.
Introduce the following APIs
```c
__subsystem struct lin_driver_api {
	lin_start_t start;
	lin_stop_t stop;
	lin_configure_t configure;
	lin_get_config_t get_config;
	lin_set_rx_filter_t set_rx_filter;
	lin_send_t send;
	lin_receive_t receive;
	lin_response_t response;
	lin_read_t read;
	lin_wakeup_send_t wakeup_send;
	lin_enter_sleep_t enter_sleep;
	lin_set_callback_t set_callback;
};
```

adamsanta, 2025-11-19:
Hello @thenguyenyf , @henrikbrixandersen , 
I am planning to program a LIN master for the NXP S32K3X4EVB-T172 board (using Zephyr of course).
I am new to Zephyr but I'd be happy to help you going on with this PR.
If I understand well, you're planning to add a `lin_renesas_ra_sci_b.c` file, based on `lin_skeleton.c`, to implement Renesas SCI specific LIN code, right ? Should I do the same for my NXP board ?
Thanks !

thenguyenyf, 2025-11-20:
Hello @adamsanta . Nice to see you!

> If I understand well, you're planning to add a lin_renesas_ra_sci_b.c file, based on lin_skeleton.c, to implement Renesas SCI specific LIN code, right ?

Yes. There are some works in background for `lin_renesas_ra_sci_b.c` since last time I pushed to https://github.com/zephyrproject-rtos/zephyr/pull/95975. `lin_renesas_ra_sci_b.c` will be sent out soon, and `lin_skeleton.c` also has some modifications with a small change in interface design.

So, if you don't mind, please wait until next weekend at the latest after I put out some updates for https://github.com/zephyrproject-rtos/zephyr/pull/95975 before starting your development.

closed 2026-09-05 by github-actions[bot]