zephyrproject-rtos/zephyr #96186: support the TI TMS320F28388D MCU?

zbqxyz, 2025-09-18:
### Is your feature request related to a problem? Please describe.

can it TI TMS320F28388D  mcu cm core ?

### Describe the solution you'd like

can it TI TMS320F28388D  mcu cm core ?

zbqxyz, 2025-09-19:
> [@zbqxyz](https://github.com/zbqxyz) sorry the information is unclear, could you give more details ?

TI's TMS320F28388D processor finds extensive applications in the field of industrial control, particularly in motor control and power electronics. This processor features 3 cores, including two DSP cores and one ARM core. Will Zephyr RTOS provide support for its architecture?

ssekar15, 2025-09-19:
> TI's TMS320F28388D processor finds extensive applications in the field of industrial control, particularly in motor control and power electronics. This processor features 3 cores, including two DSP [...]

@zbqxyz 
Currently C28x architecture is not supported in Zephyr RTOS. but the connectivity manager ARM M4 core can run Zephyr as ARM Cortex-M well supported. Adding C28x architecture support is non-trivial, but possible.

closed 2026-09-02 by github-actions[bot]