zephyrproject-rtos/zephyr #96339: In kernel.multiprocessing.smp test `test_coop_resched_threads` fails on 12-core systems due to priority range violation and insufficient timing

M-Moawad, 2025-09-21:
### Describe the bug

The `test_coop_resched_threads` test in [kernel.multiprocessing.smp]( https://github.com/zephyrproject-rtos/zephyr/blob/1c1598601b5c17f1f6e153e88ffd85c6bd5157c6/tests/kernel/smp/src/main.c) fails on 12-core systems for two reasons:

**Priority Range Violation**
The test uses [K_PRIO_COOP(10) ](https://github.com/zephyrproject-rtos/zephyr/blob/1c1598601b5c17f1f6e153e88ffd85c6bd5157c6/tests/kernel/smp/src/main.c#L375) as the starting priority, which evaluates to -6. With the default `CONFIG_NUM_COOP_PRIORITIES=16`, the allowed range: 14 to -16.

The [spawn_threads ](https://github.com/zephyrproject-rtos/zephyr/blob/1c1598601b5c17f1f6e153e88ffd85c6bd5157c6/tests/kernel/smp/src/main.c#L242) function decrements the priority for each thread created, resulting in priorities -7, -8, -9, ..., -17 for 12 threads. The thread receives priority -17, which exceeds the valid range and triggers an **invalid priority** assertion.

**Insufficient Timing**
The test uses a [fixed DELAY_US](https://github.com/zephyrproject-rtos/zephyr/blob/1c1598601b5c17f1f6e153e88ffd85c6bd5157c6/tests/kernel/smp/src/main.c#L379) busy wait period to allow threads to execute before checking results. On 12-core systems, this delay is insufficient for all threads to complete execution, causing false failures such as "Comp cooperative thread 4 didn't run."

### Regression

- [ ] This is a regression.

### Steps to reproduce

west build -p -b nsim/nsim_hs5x/smp/12cores tests/kernel/smp -T kernel.multiprocessing.smp -t run

### Relevant log output

```shell
===================================================================
START - test_coop_resched_threads
ASSERTION FAIL [((((prio)) == 15 && z_is_idle_thread_entry((entry))) || (((15 - 1) >= ((-16))) && ((prio)) >= ((-16)) && ((prio)) <= (15 - 1)))] @ WEST_TOPDIR/zephyr/kernel/thread.c:581
	invalid priority (-17); allowed range: 14 to -16
E:  r0: 0x4  r1: 0x245  r2: 0x1  r3: 0x8000a5a4
E:  r4: 0x0  r5: 0x0  r6: 0x0  r7: 0x1
E:  r8: 0xffffffff  r9: 0x0 r10: 0x1f r11: 0x0
E: r12: 0xa r13: 0x80  pc: 0x8000144e
E:  blink: 0x80006aca status32: 0x80082802
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x8000bd18 (test_coop_resched_threads)
wrong error reason
===================================================================
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

OS: Linux
Toolchain (e.g Zephyr SDK, ...): arcmwdt
Commit SHA or Version used: Main https://github.com/zephyrproject-rtos/zephyr/tree/1c1598601b5c17f1f6e153e88ffd85c6bd5157c6

M-Moawad, 2025-09-21:
The following patch makes this test pass, but both the priority values and timing parameters remain hardcoded "magic numbers" that are not dynamically calculated as a factor of the number of CPU cores.

```diff
diff --git a/tests/kernel/smp/src/main.c b/tests/kernel/smp/src/main.c
index f3d3f356eed..c1c0136c57e 100644
--- a/tests/kernel/smp/src/main.c
+++ b/tests/kernel/smp/src/main.c
@@ -19,7 +19,7 @@
 #define T2_STACK_SIZE (2048 + CONFIG_TEST_EXTRA_STACK_SIZE)
 #define STACK_SIZE (384 + CONFIG_TEST_EXTRA_STACK_SIZE)
 #define DELAY_US 50000
-#define TIMEOUT 1000
+#define TIMEOUT 5000
 #define EQUAL_PRIORITY 1
 #define TIME_SLICE_MS 500
 #define THREAD_DELAY 1
@@ -372,11 +372,11 @@ ZTEST(smp, test_coop_resched_threads)
         * since we don't give up current CPU, last thread
         * will not get scheduled
         */
-       spawn_threads(K_PRIO_COOP(10), num_threads, !EQUAL_PRIORITY,
+       spawn_threads(K_PRIO_COOP(12), num_threads, !EQUAL_PRIORITY,
                      &thread_entry_fn, THREAD_DELAY);
 
        /* Wait for some time to let other core's thread run */
-       k_busy_wait(DELAY_US);
+       k_busy_wait(DELAY_US*5);
 
 
        /* Reassure that cooperative thread's are not preempted
@@ -446,11 +446,11 @@ ZTEST(smp, test_yield_threads)
         * of cores, so the last thread would be
         * pending.
         */
-       spawn_threads(K_PRIO_COOP(10), num_threads, !EQUAL_PRIORITY,
+       spawn_threads(K_PRIO_COOP(12), num_threads, !EQUAL_PRIORITY,
                      &thread_entry_fn, !THREAD_DELAY);
 
        k_yield();
-       k_busy_wait(DELAY_US);
+       k_busy_wait(DELAY_US*5);
 
        for (int i = 0; i < num_threads; i++) {
                zassert_true(tinfo[i].executed == 1,
@@ -475,7 +475,7 @@ ZTEST(smp, test_sleep_threads)
 {
        unsigned int num_threads = arch_num_cpus();
 
-       spawn_threads(K_PRIO_COOP(10), num_threads, !EQUAL_PRIORITY,
+       spawn_threads(K_PRIO_COOP(12), num_threads, !EQUAL_PRIORITY,
                      &thread_entry_fn, !THREAD_DELAY);
 
        k_msleep(TIMEOUT);
@@ -533,7 +533,7 @@ static void check_wokeup_threads(int tnum)
        /* k_wakeup() isn't synchronous, give the other CPU time to
         * schedule them
         */
-       k_busy_wait(200000);
+       k_busy_wait(300000);
 
        for (i = 0; i < tnum; i++) {
                if (tinfo[i].executed == 1 && threads_woke_up <= tnum) {
@@ -558,7 +558,7 @@ ZTEST(smp, test_wakeup_threads)
        unsigned int num_threads = arch_num_cpus();
 
        /* Spawn threads to run on all remaining cores */
-       spawn_threads(K_PRIO_COOP(10), num_threads - 1, !EQUAL_PRIORITY,
+       spawn_threads(K_PRIO_COOP(12), num_threads - 1, !EQUAL_PRIORITY,
                      &thread_wakeup_entry, !THREAD_DELAY);
 
        /* Check if all the threads have started, then call wakeup */
```

peter-mitsis, 2025-09-22:
While it is true that the test can be adjusted to allow for 12 CPUs, it would be a band-aid solution for all that we will have done is kick the can down the road until a platform with even more CPUs comes along. That being said, we can adjust the test to work with 12 CPUs, but I also think that it will also be necessary to place a build limit within the test case.

I can imagine two ways of implementing this limit. The first would be to update testcase.yaml to filter out platforms that use more than 12 CPUs. The second (and preferable in my opinion) would be to generate a build error when more than 12 CPUs are detected. This build error would need to both explain the error and what to do about it (which could be as simple as having a custom board overlay that sets CONFIG_MP_MAX_NUM_CPUS to 12).

PR #96932 (2025-10-06): tests: Update smp test to support 12 CPUs
Updates the smp test to support nsim/nsim_hs5x/smp/12cores. Furthermore, if someone attempts to build the smp test for more than 12 CPUs, report an error message informing the user to reduce the number of CPUs. This reduction can be in the form of a custom overlay for the board in question.

Fixes #96339
closed 2025-10-06 by cfriedt