zephyrproject-rtos/zephyr #96389: Linking with lld generates warnings

tpambor, 2025-09-22:
### Describe the bug

Compiling with LLVM as part of zephyr sdk 0.18.0-alpha4, lld generates warnings. These warnings have previously been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/71546. It seems that `CONFIG_LLVM_USE_LLD` is not set even though `lld` is used to link.

### Regression

- [x] This is a regression.

### Steps to reproduce

1. `cd samples/hello_world`
2. `west build -b stm32h573i_dk -- -DZEPHYR_TOOLCHAIN_VARIANT=zephyr/llvm`

### Relevant log output

```shell
ld.lld: warning: <internal>:(.symtab) is being placed in '.symtab'
ld.lld: warning: <internal>:(.shstrtab) is being placed in '.shstrtab'
ld.lld: warning: <internal>:(.strtab) is being placed in '.strtab'
```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

### Environment

- OS: Linux
- Toolchain: Zephyr SDK v0.18.0-alpha4
- Version: collab-sdk-dev

tpambor, 2025-09-22:
/cc @stephanosio

tpambor, 2026-09-02:
I submitted a fix to the SDK https://github.com/zephyrproject-rtos/sdk-ng/pull/1162

onaffaa, 2026-11-09:
Regarding this fix, is there any benefit to not using `__LLD_LINKER_CMD__`, which would negate the need for `CONFIG_LLVM_USE_LLD` entirely since it is defined when Zephyr preprocesses the linker script and determines that the linker being used is LLVM LLD.

I ran into and proposed something similar to #120369 in #125005, and after being made aware of this fix my opinion is only one of the 2 symbols need to be used. If I did have to pick I would lean towards the usage of `__LLD_LINKER_CMD__` but I'm open to either solution.