zephyrproject-rtos/zephyr #96498: [NXP][MIMXRT1180-EVK]: Network stack problem on CM7: Error on received frame

javad123javad, 2025-09-24:
### Describe the bug

Hello,
When running the application on the `MIMXRT1180-EVK` (`CM7` core), the following error appears in the console in an infinite loop

It originates from:
drivers/ethernet/nxp_imx_netc/eth_nxp_imx_netc.c

```bash
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
E: Error on received frame
```

The following project configuration is used:

```bash
# General config
CONFIG_HEAP_MEM_POOL_SIZE=2097152
CONFIG_MAIN_STACK_SIZE=20336
CONFIG_SHELL_STACK_SIZE=8192
CONFIG_POSIX_API=y
CONFIG_POSIX_TIMERS=y
CONFIG_POSIX_C_LIB_EXT=y
CONFIG_SOCKS=y
CONFIG_INIT_STACKS=y
CONFIG_YAML=y
CONFIG_POSIX_C_LANG_SUPPORT_R=y
# Networking config

CONFIG_NETWORKING=y   				
CONFIG_NET_INTERFACE_NAME=y         
CONFIG_NET_IPV4=y					
CONFIG_NET_IPV6=n                   
CONFIG_NET_IPV4_AUTO=n				
CONFIG_NET_TCP=y					
CONFIG_NET_UDP=y
CONFIG_NET_SOCKETS=y				
CONFIG_ZVFS_POLL_MAX=32
CONFIG_ZVFS_OPEN_MAX=32

CONFIG_NET_DRIVERS=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_LOOPBACK=y				
CONFIG_NET_MGMT=y
CONFIG_NET_MGMT_EVENT=y
CONFIG_DNS_RESOLVER=y

CONFIG_NET_DEFAULT_IF_ETHERNET=y
CONFIG_NET_MAX_CONN=16
CONFIG_NET_MAX_CONTEXTS=16
CONFIG_NET_TX_STACK_SIZE=1500

CONFIG_NET_IF_MCAST_IPV4_ADDR_COUNT=8
CONFIG_NET_BUF_DATA_SIZE=4096

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_NEED_IPV4=y

CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.168.100.21"

#CONFIG_NET_DHCPV4=y

CONFIG_NET_CONFIG_INIT_TIMEOUT=120
CONFIG_NET_NATIVE=y

CONFIG_NET_LOG=y
CONFIG_NET_TCP_LOG_LEVEL_WRN=y
CONFIG_NET_CONTEXT_LOG_LEVEL_WRN=y
CONFIG_NET_SOCKETS_LOG_LEVEL_WRN=y

```

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

DerekSnell, 2025-09-24:
@yangbolu1991 , FYI
Would some Wireshark captures of these frames be helpful?

yangbolu1991, 2025-09-25:
@javad123javad ,
It may be related to PHY link stability between USB–LAN adapter and your EVK board. 
Could you try to connect EVK to other Ethernet device? Or, change to use another EVK to connect other Ethernet device?

Also the Ethernet device should support negotiation to 100M speed.
Thanks.

dpiskula-nxp, 2025-09-25:
Hi @javad123javad, thank you very much for the additional context. You are right, the error will need a more graceful handling. It seems the function can only break out of the loop if it receives -ENOBUFS but in your case it most likely receives -EIO instead. Would you be able to pinpoint where the -EIO is returned from by the netc_eth_rx() function in eth_nxp_imx_netc.c? I assume it will be either due to the if at line 99 or at line 109. This could further help us find the root cause while we work on a patch with a more graceful handling of the error.

Additionally, if you'd like to try getting out of the infinite loop, you could try modifying the file from line 172 to be:

```
int rx_result = netc_eth_rx(dev);
while ((rx_result != -ENOBUFS) && (rx_result != -EIO)) {
    if (++work == CONFIG_ETH_NXP_IMX_RX_BUDGET) {
        /* more work to do, reschedule */
        work = 0;
        k_yield();
    }
    rx_result = netc_eth_rx(dev);
}
```
Though, it's hard to tell how the application will behave afterwards.

javad123javad, 2025-09-25:
> Hi [@javad123javad](https://github.com/javad123javad), thank you very much for the additional context. You are right, the error will need a more graceful handling. It seems the function can only [...]

Hi @dpiskula-nxp ,
Thanks a lot for the hint. 
Unfortunately, I'm not sure for the moment from which function it is returned. I will modify the driver file to print the line number beside the error message to see where exactly it is happened.

javad123javad, 2025-10-01:
Hi, @dpiskula-nxp 
The frame error occurred again. After modifying the code to check the return value, it no longer enters an infinite loop. However, I think the network interface is still not functioning properly, as I can no longer ping the peer host (my PC).

Here’s where it happened:

```c

```if (result != kStatus_NETC_RxTsrResp &&
            result != kStatus_NETC_RxHRNotZeroFrame &&
            result != kStatus_Success) {
                LOG_ERR("Error on received frame: Line: %d", __LINE__);
                ret = -EIO;
                goto out;
        }
```

dpiskula-nxp, 2025-10-03:
Hi @javad123javad, thanks for the update. Can you use Wireshark or something similar to monitor what is happening? Since we have no way of reproducing the problem on our side, we need much more detailed info to have any chance of figuring out what's wrong.

Do you think the cable length is the root cause? It could be interesting to see how the communication looks when using the shorter cable with both with this RT1180 device and with whatever you tested where it works and then compare them with the longer cable that works with RT1180. Maybe we'll be able to see some issues that don't matter to other devices but for some reason cause errors on RT1180.

javad123javad, 2025-10-03:
> Hi [@javad123javad](https://github.com/javad123javad), thanks for the update. Can you use Wireshark or something similar to monitor what is happening? Since we have no way of reproducing the problem [...]

Hi @dpiskula-nxp 
Thank you very much for the hint. I completely understand that more information is needed. It’s also been a bit challenging on our side to reproduce the issue. However, we have disabled the network stack on CM33. Additionally, on the CM7 side, we disabled the unused network ifaces. After these changes, we no longer observed the problem (both up/down and also frame error).

In any case, I will keep an eye on it, to see if it happens again or not.

Kind regards,
Javad

javad123javad, 2025-10-06:
Hi @dpiskula-nxp , 
I will close this issue, as we didn't see it to happen by aforementioned changes. If it happened again, I will open a new one.

closed 2025-10-06 by javad123javad

ardaotuz, 2026-11-06:
This is still present on main (f3fc249035296b87c75fa808b3abf78829fe9378), see also #100037.

When `EP_GetRxFrameSize()` returns an error such as `kStatus_NETC_RxFrameError`, the RX path returns `-EIO` without consuming the errored descriptor, and the RX thread's loop retries the same descriptor:

- https://github.com/zephyrproject-rtos/zephyr/blob/f3fc249035296b87c75fa808b3abf78829fe9378/drivers/ethernet/nxp_imx_netc/eth_nxp_imx_netc.c#L134-L145
- https://github.com/zephyrproject-rtos/zephyr/blob/f3fc249035296b87c75fa808b3abf78829fe9378/drivers/ethernet/nxp_imx_netc/eth_nxp_imx_netc.c#L221-L235

The RX thread then keeps running at its cooperative priority and starves lower-priority threads. Dropping the errored frame (e.g. `EP_ReceiveFrameCopy(handle, 0, NULL, 0, NULL)`) and counting it as an RX error would let the ring advance. Could this issue be reopened until a fix lands?