zephyrproject-rtos/zephyr #96560: kernel: mem_slab: k_mem_slab_free does not check for empty mem slab

arnebdr, 2025-09-25:
### Describe the bug

The function k_mem_slab_free does not check if the memory slab is empty before removing an element.
This results in "num_used" rolling over to 0xFFFFFFFF when free is called with a valid pointer too often in this line:
https://github.com/zephyrproject-rtos/zephyr/blob/376ca571b08e4c8795da32290d60fa0e72958795/kernel/mem_slab.c#L296

### Regression

- [ ] This is a regression.

### Relevant log output

```shell

```

### Impact

Annoyance – Minor irritation; no significant impact on usability or functionality.

arnebdr, 2025-10-02:
Hi @ljd42 ,
as example for this I use the hello_world sample and replace the main.c code by the code below.
It allocates a buffer and then frees it twice.
The num_used counter rolls over after the second free. 
A second alloc is successfull but the num_used counter is 0 afterwards.
CONFIG_ASSERT does not change anything.

The output of the program is:
> First alloc ok
> Mem slab num used after alloc: 1
> Mem slab num used after first free: 0
> Mem slab num used after second free: 4294967295
> Second alloc ok
> Mem slab num used after second alloc: 0

```
#include <stdio.h>
#include <zephyr/kernel.h>

#define BLOCK_SIZE_TX 4096  // size in bytes
#define NUM_BLOCKS_TX 6

K_MEM_SLAB_DEFINE_STATIC(txSlab, BLOCK_SIZE_TX, NUM_BLOCKS_TX, 4);

int main(void)
{
	void* txBlock;
	int err;
	
	err = k_mem_slab_alloc(&txSlab, &txBlock, K_NO_WAIT);
    if (err < 0)
    {
        printk("Failed to allocate mem slab\n");
        return err;
    }
	else {
		printk("First alloc ok\n");
		
		printk("Mem slab num used after alloc: %u\n", k_mem_slab_num_used_get(&txSlab));
		
		k_mem_slab_free(&txSlab, txBlock);
		printk("Mem slab num used after first free: %u\n", k_mem_slab_num_used_get(&txSlab));
		
		k_mem_slab_free(&txSlab, txBlock);
		printk("Mem slab num used after second free: %u\n", k_mem_slab_num_used_get(&txSlab));
	}
	
	err = k_mem_slab_alloc(&txSlab, &txBlock, K_NO_WAIT);
    if (err < 0)
    {
        printk("Second alloc failed\n");
    }
	else {
		printk("Second alloc ok\n");
		printk("Mem slab num used after second alloc: %u\n", k_mem_slab_num_used_get(&txSlab));
	}
	
	return 0;
}
```

ljd42, 2025-10-07:
Doing a double free is extremely problematic. It's not only a wrong "number of slab used". In fact, it corrupts the `free_list` and therefore the entire  container. In your example, you just happened to free the last slab  twice, so nothing too bad is happening. A more severe scenario would be to free k slabs 1,2,3,..k in that order and then free slab 1 again.

I can imagine case where one needs more "safety". Of course, that cannot be at the expenses of every fw/app - especially those that need performance and don't have any double free bug. But that tension can be easily resolved.

I think I found a nice O(1) implementation to detect double free. But that led me to consider one thing. I can easily reset the slab after freeing. Then a double free would be  detected with the previous method: 
```c 
k_mem_slab_free(&txSlab, txBlock);
txBlock=NULL;
k_mem_slab_free(&txSlab, txBlock); // assert fails here
```

@arnebdr : Could the above method be easily applied to your application? Could you shed some lights about your use cases for having a "safe" k_mem_slab_free() ? I am happy to propose a solution. Just checking first if it can be achieved in some simple way :wink:

cfriedt, 2025-10-08:
> I just wondered why the free method is not checking for an empty slab before it frees the memory.
> This check would be easy and the performance impact is minimal.
> But if this was left out by intention, I can live with it.

@arnebdr - it would be best to add the check, I think. I might have missed if there is already a pull request to fix it. If there is not, would you be able to make one?