zephyrproject-rtos/zephyr #96761: dts/edt: Missing fixed-partition and fixed-subpartition compatibles in pickled output

nordicjm, 2025-09-30:
### Describe the bug

There seems to be an issue likely in one of the following files:
`scripts/dts/gen_dts_cmake.py`
`scripts/dts/gen_edt.py`
`scripts/dts/python-devicetree/src/devicetree/edtlib.py`

When an input dts file with `fixed-partitions` or `fixed-subpartitions` is used, the output pickled file does not contain that this DT compatible has been used, I had a look and somewhere in there it seems like the `compats` property is set in the node object with the valid compatible but there is no `compatible` property in the node object, I was unable to work out where or how that was set in the code

Without this working, it is not possible to use CMake DT functions to be able to e.g. list what partitions are using the `fixed-subpartition` compatible vs the `fixed-partition` one

### Regression

- [ ] This is a regression.

### Steps to reproduce

Build for a board e.g. `nrf9160dk/nrf9160` and check pickled output file, there is no `DT_COMP` line for either `fixed-subpartitions` or `fixed-partition`

### Relevant log output

```shell

```

### Impact

Major – Severely degrades functionality; workaround is difficult or unavailable.

### Environment

Commit: d90d71c42c6d3a81b10b17b5eb5ab3d686b7512f

mbolivar, 2025-09-30:
@nordicjm not sure the problem is in edtlib.py. For example:

```
$ west build -b nrf9160dk/nrf9160 --cmake-only samples/hello_world/
```

With this, I can see 'fixed-partitions' in the EDT object's compat2nodes, for example:

```
(zp) mbolivar@MBOLIVAR:~/src/zp/zephyr (main)$ cd build/zephyr
(zp) mbolivar@MBOLIVAR:~/src/zp/zephyr/build/zephyr (main)$ ipython
Python 3.12.3 (main, Aug 14 2025, 17:47:21) [GCC 13.3.0]
Type 'copyright', 'credits' or 'license' for more information
IPython 9.6.0 -- An enhanced Interactive Python. Type '?' for help.
Tip: You can use Ctrl-O to force a new line in terminal IPython

In [1]: import pickle

In [2]: edt= pickle.load(open('edt.pickle', 'rb'))

In [3]: edt.compat2nodes.keys()
Out[3]: dict_keys(['nordic,nrf9160-dk-nrf9160', 'nordic,nrf9160-sica', 'nordic,nrf9160', 'nordic,nrf91', 'simple-bus', 'arm,v8m-nvic', 'mmio-sram', 'nordic,nrf91-flash-controller', 'soc-nv-flash', 'fixed-partitions', 'fixed-subpartitions', 'nordic,nrf-saadc', 'nordic,nrf-dppic', 'nordic,nrf-egu', 'nordic,nrf-ipc', 'nordic,nrf-kmu', 'nordic,nrf91x-regulators', 'nordic,nrf-vmc', 'nordic,nrf-uarte', 'nordic,nrf-twim', 'nordic,nrf-spim', 'nordic,nrf-pwm', 'nordic,nrf-gpio', 'nordic,nrf-clock', 'nordic,nrf-power', 'nordic,nrf-gpregret', 'nordic,nrf-wdt', 'nordic,nrf-ctrlapperi', 'nordic,nrf-gpiote', 'nordic,nrf-spu', 'nordic,nrf-ficr', 'nordic,nrf-uicr',
'nordic,nrf-pinctrl', 'arm,cortex-m33f', 'arm,armv8m-mpu', 'gpio-leds', 'pwm-leds', 'gpio-keys', 'nordic,nrf9160dk-nrf52840-interface', 'arduino-header-r3', 'arduino,uno-adc', 'arm,armv8m-systick', 'nordic,nrf-i2s', 'nordic,nrf-pdm', 'nxp,pcal6408a', 'jedec,spi-nor', 'nordic,nrf-rtc', 'nordic,nrf-timer', 'arm,cryptocell-310', 'zephyr,bt-hci-entropy', 'nordic,nrf-sw-pwm', 'nordic,nrf9160dk-nrf52840-reset'])
```

Please clarify exactly what's missing in the python part of this. I don't see it missing.

nordicjm, 2025-10-01:
> [@nordicjm](https://github.com/nordicjm) - is there any other repro information available. Is the error something we should expect to see in CI?

It should be visible in any build that uses a board with partitions in dts, but nothing as of yet in the upstream zephyr tree makes use of searching for this compatible

mbolivar, 2025-10-01:
OK, so the problem does not seem to be in the pickled output, but rather in the cmake output. Changing the title appropriately

mbolivar, 2025-10-01:
I've found the problematic code, working on a fix

closed 2025-10-04 by cfriedt